m-labs/misoc #4: LiteEth: Add support for Digilent Atlys board

mithro, 2015-03-17:
(I'm logging this bug for advice on the correct way to implement this and so other people can see the request.)

The Digilent Atlys board is a Spartan 6 prototyping board with a Marvell Alaska Tri-mode PHY (the 88E1111) paired with a Halo HFJ11-1G01E RJ-45 connector. 

From the [Atlys reference manual](http://www.digilentinc.com/Data/Products/ATLYS/Atlys_rm_V2.pdf), it says
> Both MII and GMII interface modes are supported at 10/100/1000 Mb/s. 

Sadly it also says;
> The data sheet for the Marvell PHY is available from Marvell only with a valid NDA. Please contact
Marvell for more PHY-specific information.

The fact that it supports the MII and GMII interface and you seem to have support at https://github.com/enjoy-digital/liteeth/blob/master/liteeth/phy/gmii.py - would it just be a matter of connecting it up? As well;
> EDK-based designs can access the PHY using either the xps_ethernetlite IP core for 10/100 Mbps
designs, or the xps_ll_temac IP core for 10/100/1000 Mbps designs. 

More useful information seems to be;
> Default settings used at power-on or reset are:
>  * MII/GMII mode to copper interface
>  * Auto Negotiation Enabled, advertising all speeds, preferring Slave
>  * MDIO interface selected, PHY MDIO address = 00111
>  * No asymmetric pause, no MAC pause, automatic crossover enabled
>  * Energy detect on cable disabled (Sleep Mode disabled), interrupt polarity LOW

Here is a wiring diagram;
[image]

BTW We had a student work on using the ethernet interface directly last year. The result of his work can be found at https://docs.google.com/document/d/1PSjfm6eS0B3UUPJmPf7PH0tNsF7ZFKIKfPldmF3ucKY/edit and http://hdmi2ethernet.blogspot.com.au/

(Copied from original https://github.com/enjoy-digital/liteeth/issues/2 at @enjoy-digital 's request.)

enjoy-digital, 2015-03-18:
The board KC705 we are currently using has the same Marvel 88E1111 PHY. So if you want to use LiteEth with Migen/MiSoC, the only thing you have to do is to declare the ethernet pins in your platform file as it's done here : https://github.com/m-labs/migen/blob/master/mibuild/platforms/kc705.py#L128

You can then just try a LiteEth example design:
- go to misoclib/com/liteeth/example_designs
- ./make.py -t base -p your_platform make all
and try to ping your board :)

Please tell us in case you try it and the results you have.

Thanks for your hdmi2ethernet link, if you want to use LiteEth with your design, you can just plug directly your RTP layer to the UDP layer of LiteEth (and remove the IP/UDP stuff you added to your RTP). LiteEth will then handle the UDP/IP/ARP/ICMP/MAC layers.

enjoy-digital, 2015-04-17:
I'm closing this since the Altys board uses the same PHY we already support on the KC705. BTW we added dynamic MII / GMII switching recently so you can use the same bitstream for 10/100/1000Mbps.

mithro, 2015-04-20:
Can we keep this issue open until I finish adding the Atlys board to misoc? At the moment nobody has actually tested that the LiteEth works with the Atlys board.

enjoy-digital, 2015-04-20:
Np. Please share your results when it will be tested.

enjoy-digital, 2015-07-02:
Validated on Altys board with 10/100/1000 Mbps triple mode PHY:
https://github.com/timvideos/HDMI2USB-misoc-firmware

closed 2015-07-02 by enjoy-digital