m-labs/misoc #5: LiteEth: Add support for RGMII phy interface

mithro, 2015-03-17:
(I'm logging this bug for advice on the correct way to implement this and so other people can see the request.)

It looks like you already support the GMII interface at https://github.com/enjoy-digital/liteeth/blob/master/liteeth/phy/gmii.py but we need to support RGMII ICs.

The [RGMII interface](http://en.wikipedia.org/wiki/Media_Independent_Interface#Reduced_Gigabit_Media_Independent_Interface) reduces the number of pins needed from 24 pins for GMII to 12 pins. This reduction is achieved by clocking data on both the rising and falling edges of the clock.

From wikipedia;
> RGMII uses half the number of data pins as used in the GMII interface. This reduction is achieved by clocking data on both the rising and falling edges of the clock in 1000 Mbit/s operation, and by eliminating non-essential signals (carrier-sense and collision-indication). Thus RGMII consists only of: RX_CTL, RXC, RXD[3:0], TX_CTL, TXC, TXD[3:0] (12 pins, as opposed to GMII's 24).
>
> Unlike GMII, the transmit clock signal is always provided by the MAC on the TXC line, rather than being provided by the PHY for 10/100 Mbit/s operation and by the MAC at 1000 Mbit/s. Source-synchronous clocking is used: the clock signal that is output (by either the PHY or the MAC) is synchronous with the data signals. This requires the PCB to be designed to add a 1.5-2ns delay to the clock signal in order to make the setup and hold times on the sink. RGMII v2.0 specifies an optional internal delay, obviating the need for the PCB designer to add delay; this is known as RGMII-ID.

What is the best way to go about adding this?

(Copied from original https://github.com/enjoy-digital/liteeth/issues/1 at @enjoy-digital 's request.)

enjoy-digital, 2015-03-18:
Yes that's something we plan to add and that would also be a nice contribution.
It should be very similar to GMII with only DDR support to input/output data.

If you want to add it, create your rgmii.py file, use GMII as a starting point and add the DDR stuff to it (IDDR2/ODDR2 on Xilinx devices). Also please try to indicate in the code where it's vendor specific, and where it's generic so that it will ease contribution for others FPGA vendors.

If you do it, please send us your rgmii.py as a patch, we will review it and integrate it.

mithro, 2015-09-24:
I'm pretty sure this has been done by @enjoy-digital (https://github.com/enjoy-digital/liteeth/blob/master/liteeth/phy/s6rgmii.py)
liteEth has been moved into a separate repo again (https://github.com/enjoy-digital/liteeth/blob/master)  as part of the reorg in https://github.com/m-labs/misoc/issues/17

closed 2015-09-24 by mithro

mithro, 2015-09-24:
Sorry, that link to liteeth should be https://github.com/enjoy-digital/liteeth