m-labs/misoc #6: First step in quick start guide doesn't link to further setup instructions for migen

mithro, 2015-04-13:
I'm just trying to follow the "[> Quick start guide" guide from the README.md at https://github.com/m-labs/misoc

First step is listed as;
> 1. Install Python 3.3+, Migen and FPGA vendor's development tools.
>   Get Migen from: https://github.com/m-labs/migen

I clicked the link to migen and it is very unclear on how I should be installing migen. There is no "migen install instructions" at https://github.com/m-labs/migen

This leads me to a couple of questions;
 * Could the Migen link to a better "migen install page"?
 * Does such a "Migen install page" exist?
 * Is "Migen" just a python module? Could it be installed in a virtualenv or via pip?

It looks like there is lot of useful instructions at http://m-labs.hk/artiq/manual/installing.html and http://m-labs.hk/gateware.html - specially the http://m-labs.hk/migen-tutorial.pdf document.

fallen, 2015-04-13:
Indeed, Migen is a python module, you can then use pip or any other tool able to process setuptools meta data (in setup.py).
If you want to contribute, then the best way to install it is using : ``python3 ./setup.py develop --user``

There is a documentation for Migen, which contains an "install" instruction : https://migen.readthedocs.org/en/latest/introduction.html#installing-migen

It states that for simulation, you also need to install the VPI module.

Migen can also be installed via conda : https://binstar.org/fallen/migen
However, as the migen conda package is not update automatically yet, it can be outdated and you should prefer the "install from git clone" solution.

closed 2015-04-14 by sbourdeauducq

mithro, 2015-04-19:
There currently still isn't a clear path from the README.md in the misoc directory to the install instructions at https://migen.readthedocs.org/en/latest/introduction.html#installing-migen

See https://github.com/m-labs/migen/issues/13