m-labs/misoc #12: Remove one of the two flterm versions

sbourdeauducq, 2015-05-07:
If removing the C version, clean up the python version (e.g. no threads).

enjoy-digital, 2015-05-09:
OK, I'll look at that, you suggest asyncio to remove threads?

sbourdeauducq, 2015-05-09:
Yes, though asyncio won't work easily with pyserial yet.

sbourdeauducq, 2015-10-19:
Done in new branch. We are keeping only the Python version, when asyncserial works we should get rid of the threads.

sbourdeauducq, 2016-03-02:
asyncio flterm is in separate branch. Works on Linux, needs testing on Windows.

enjoy-digital, 2016-03-04:
Just tested flterm on windows, it seems to freeze here:

https://github.com/m-labs/misoc/blob/asyncserial/misoc/tools/flterm.py#L201
https://github.com/m-labs/pyserial/blob/master/serial/aio.py#L121
https://github.com/python/asyncio/blob/master/asyncio/windows_events.py#L425
https://github.com/python/asyncio/blob/master/asyncio/windows_events.py#L605

sbourdeauducq, 2016-05-08:
Works now.

closed 2016-05-08 by sbourdeauducq