m-labs/misoc #13: Remove litescope references from UART

sbourdeauducq, 2015-05-09:
And do not call the WB bridge "frontend".

enjoy-digital, 2015-05-09:
I've tried to change this, hope it's better.

sbourdeauducq, 2015-05-09:
I don't think the name ``uart.wishbone`` is relevant. Those names should reflect the role that the UART has, either as a target device that receives the characters to write from a CPU through a bus (Wishbone, CSR, ...), or as an initiator device that masters a bus.

sbourdeauducq, 2015-05-09:
Maybe ``uart.UART`` for the normal CSR target peripheral (since this is the most common, default, assumed role of a UART in a SoC), and ``uart.bridge.Wishbone`` for the initiator/bridge.

enjoy-digital, 2015-05-09:
Thanks, so is it OK if I rename wishbone.py in bridge.py ? (I want to avoid creating a bridge directory for now since we only have a wishbone bridge, eventually later if we have others bridges (AXI, Avalon))

sbourdeauducq, 2015-05-09:
Sounds good.

enjoy-digital, 2015-05-09:
Done, thanks for the feedback!

sbourdeauducq, 2015-05-09:
Thanks for fixing it!

closed 2015-05-09 by sbourdeauducq