m-labs/misoc #19: misoc should put all temporary and build assets in a separate directory

mithro, 2015-09-24:
This will probably happen as part of https://github.com/m-labs/misoc/issues/17 (as if misoc is installed into the root of the system, it won't be able to write to the directory it is installed into) but thought I would specifically call it out in a separate issue.

At the moment it is my understanding that misoc creates the following things;
 * generated header files - This currently end up in `misoc/software/include/generated/`
 * compiler objects - This currently end up in C library directories `misoc/software/xxxx`
 * xilinx output - This currently ends up in `misoc/build`

Did I miss anything?

sbourdeauducq, 2015-10-19:
New branch does this.

closed 2015-10-19 by sbourdeauducq

mithro, 2015-10-19:
Awesome, thanks!