m-labs/misoc #23: new misoc: building with lm32 fails on windows

enjoy-digital, 2015-10-18:
ERROR:HDLCompiler:281 - "C:\Python34\lib\site-packages\misoc-1.0-py3.4.egg\misoc
\cores\lm32\verilog\submodule\rtl\/lm32_include.v" Line 61: Cannot open include
file "lm32_config.v".

enjoy-digital, 2015-10-19:
Similar issue with Vivado:
```text
ERROR: [Synth 8-1766] cannot open include file lm32_config.v [C:/Python34/lib/si
te-packages/misoc-1.0-py3.4.egg/misoc/cores/lm32/verilog/submodule/rtl/lm32_incl
ude.v:61]

sbourdeauducq, 2015-10-19:
Does the file exist:
C:/Python34/lib/site-packages/misoc-1.0-py3.4.egg/misoc/cores/lm32/verilog/submodule/rtl/lm32_include.v

enjoy-digital, 2015-10-19:
In fact that's lm32_config.v that is missing in C:/Python34/lib/site-packages/misoc-1.0-py3.4.egg/misoc/cores/lm32/verilog.

It works fine when putting it manually there.

sbourdeauducq, 2015-10-19:
Ok. Both with ISE and Vivado?

sbourdeauducq, 2015-10-19:
I actually cannot test setup.py because python 3.5 or the version of setuptools that comes with it in the arch linux packages borked the installation of data directories...

enjoy-digital, 2015-10-19:
It's fine with ISE & Vivado yes. I can look at it, where is the core responsible for the copy of lm32_config.v?

sbourdeauducq, 2015-10-19:
766b0bee655ad0dda2bce920c051b3a78012cf13 should fix the problem.

enjoy-digital, 2015-10-19:
Thanks, fixed!

closed 2015-10-19 by enjoy-digital

sbourdeauducq, 2015-10-19:
:+1: Thanks for testing!