m-labs/misoc #24: new misoc: building with lm32 fails on windows with Vivado

enjoy-digital, 2015-10-18:
```text
****** Vivado v2015.1 (64-bit)
  **** SW Build 1215546 on Mon Apr 27 19:22:08 MDT 2015
  **** IP Build 1209967 on Tue Apr 21 11:39:20 MDT 2015
    ** Copyright 1986-2015 Xilinx, Inc. All Rights Reserved.

source top.tcl
# add_files C:\Python34\lib\site-packages\misoc-1.0-py3.4.egg\misoc\cores\lm32\v
erilog\submodule\rtl\lm32_icache.v
ERROR: [Vivado 12-172] File or Directory 'C:Python34libsite-packagesmisoc-1.0-py
tllm32_icache.v' does not existmodule
INFO: [Common 17-206] Exiting Vivado at Sun Oct 18 22:10:53 2015...
```
and probably the same issue than #23

sbourdeauducq, 2015-10-19:
m-labs/migen@ec80f0fa7e902fc7d5bfef1e76612564d5781927 should fix this, can you test?
It still builds OK on Linux.

enjoy-digital, 2015-10-19:
Thanks, works fine now.
I closing this an only keep one issue for the problem with the "lm32_config.v" include.

closed 2015-10-19 by enjoy-digital