m-labs/misoc #28: flterm broken when run under SSH

whitequark, 2016-02-25:
```
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/home/whitequark/miniconda/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/home/whitequark/miniconda/lib/python3.5/threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "flterm.py", line 242, in writer
    b = getkey()
  File "flterm.py", line 19, in getkey
    old = termios.tcgetattr(fd)
termios.error: (25, 'Inappropriate ioctl for device')
```

sbourdeauducq, 2016-03-02:
Works fine here. Are you sure the culprit isn't e.g. tmux or mosh?

whitequark, 2016-03-02:
Sorry, was unclear. By "under SSH" I meant like `ssh flterm ...`.

sbourdeauducq, 2016-03-02:
Does ssh have a good reason for disallowing termios calls on the pty (or pipe?) it gives to the command you run? If you don't do those termios calls, the behavior becomes funny (double echo, characters only sent after you press ENTER).

Does ``ssh -t`` help?

whitequark, 2016-03-02:
Yes, `ssh -t` helps. I guess this is a solution.

closed 2016-03-02 by whitequark

mithro, 2016-03-02:
flterm should probably detect the lack of pty and throw a more useful error?