m-labs/misoc #32: ^C shouldn't crash flterm

whitequark, 2016-03-02:
Right now:
```
[FLTERM] Starting....
Traceback (most recent call last):
  File "miniconda/bin/flterm", line 9, in <module>
    load_entry_point('misoc==0.1', 'console_scripts', 'flterm')()
  File "/home/whitequark/miniconda/lib/python3.5/site-packages/misoc/tools/flterm.py", line 278, in main
    flterm.join(True)
  File "/home/whitequark/miniconda/lib/python3.5/site-packages/misoc/tools/flterm.py", line 256, in join
    self.writer_thread.join()
  File "/home/whitequark/miniconda/lib/python3.5/threading.py", line 1054, in join
    self._wait_for_tstate_lock()
  File "/home/whitequark/miniconda/lib/python3.5/threading.py", line 1070, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt
```

I have already fixed this once.

sbourdeauducq, 2016-03-02:
Try the asyncio version. The threaded one will go, since threads are impossible to do right in Python.

sbourdeauducq, 2016-03-02:
You also have the C version available on the buildserver.

closed 2016-03-02 by sbourdeauducq