m-labs/misoc #35: Usage of symlinks on Windows is not convenient

enjoy-digital, 2016-03-04:
On windows, we need to run the console as administrator to use symlinks which is very un-convenient.
Can I change this: https://github.com/m-labs/misoc/blob/master/misoc/integration/builder.py#L110
with this:
````python
if sys.platform == "win32":
    # using symlinks on Windows is not convenient (console
    # needs to be run as administrator), copy Makefile instead.
    shutil.copy(src, dst)
else:
    os.symlink(src, dst)

jordens, 2016-03-04:
Can't you just `make -C dst_dir -f src_dir/Makefile`?

enjoy-digital, 2016-03-04:
Indeed this seems to work, thanks! (I'll validate that on Windows and Linux in the next days and will commit it)

enjoy-digital, 2016-03-04:
fixed via https://github.com/m-labs/misoc/commit/74a55257bef8767b5c9908dab0b073b728b9945c

closed 2016-03-04 by enjoy-digital