m-labs/misoc #50: setup.py fails on Ubuntu 16.04

John-K, 2017-03-21:
binutils and gcc have been cross-compiled and installed, python is 3.5.2

It seems some python files included are python2 syntax?

~/misoc$ sudo python3 setup.py install
/usr/lib/python3/dist-packages/setuptools/dist.py:285: UserWarning: Normalizing '0.6.dev' to '0.6.dev0'
  normalized_version,
running install
Checking .pth file support in /usr/local/lib/python3.5/dist-packages/
/usr/bin/python3 -E -c pass
TEST PASSED: /usr/local/lib/python3.5/dist-packages/ appears to support .pth files
running bdist_egg
running egg_info
writing dependency_links to misoc.egg-info/dependency_links.txt
writing misoc.egg-info/PKG-INFO
writing requirements to misoc.egg-info/requires.txt
writing top-level names to misoc.egg-info/top_level.txt
writing entry points to misoc.egg-info/entry_points.txt
reading manifest file 'misoc.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'misoc.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
byte-compiling build/bdist.linux-x86_64/egg/misoc/software/compiler_rt/lib/asan/scripts/asan_symbolize.py to asan_symbolize.cpython-35.pyc
  File "build/bdist.linux-x86_64/egg/misoc/software/compiler_rt/lib/asan/scripts/asan_symbolize.py", line 87
    print ' '.join(cmd)
            ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-x86_64/egg/misoc/software/compiler_rt/lib/dfsan/scripts/build-libc-list.py to build-libc-list.cpython-35.pyc
  File "build/bdist.linux-x86_64/egg/misoc/software/compiler_rt/lib/dfsan/scripts/build-libc-list.py", line 96
    print 'fun:%s=uninstrumented' % f
                                ^
SyntaxError: Missing parentheses in call to 'print'

byte-compiling build/bdist.linux-x86_64/egg/misoc/software/compiler_rt/lib/sanitizer_common/scripts/sancov.py to sancov.cpython-35.pyc
  File "build/bdist.linux-x86_64/egg/misoc/software/compiler_rt/lib/sanitizer_common/scripts/sancov.py", line 87
    print "0x%x" % i
               ^
SyntaxError: Missing parentheses in call to 'print'

byte-compiling build/bdist.linux-x86_64/egg/misoc/software/compiler_rt/test/asan/android_commands/android_common.py to android_common.cpython-35.pyc
  File "build/bdist.linux-x86_64/egg/misoc/software/compiler_rt/test/asan/android_commands/android_common.py", line 13
    print args
             ^
SyntaxError: Missing parentheses in call to 'print'

byte-compiling build/bdist.linux-x86_64/egg/misoc/software/compiler_rt/test/asan/android_commands/android_compile.py to android_compile.cpython-35.pyc
  File "build/bdist.linux-x86_64/egg/misoc/software/compiler_rt/test/asan/android_commands/android_compile.py", line 24
    print "No output file name!"
                               ^
SyntaxError: Missing parentheses in call to 'print'

copying misoc.egg-info/PKG-INFO -> build/bdist.linux-x86_64/egg/EGG-INFO
copying misoc.egg-info/SOURCES.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying misoc.egg-info/dependency_links.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying misoc.egg-info/entry_points.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying misoc.egg-info/requires.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying misoc.egg-info/top_level.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
writing build/bdist.linux-x86_64/egg/EGG-INFO/native_libs.txt
zip_safe flag not set; analyzing archive contents...
python3.5.__pycache__.site.cpython-35: module references __file__
python3.5.__pycache__.warnings.cpython-35: module references __file__
python3.5.encodings.__pycache__.__init__.cpython-35: module references __file__
python3.5.importlib.__pycache__.__init__.cpython-35: module references __file__
python3.5.importlib.__pycache__.__init__.cpython-35: module references __path__
python3.5.importlib.__pycache__._bootstrap.cpython-35: module references __file__
python3.5.importlib.__pycache__._bootstrap.cpython-35: module references __path__
python3.5.importlib.__pycache__._bootstrap_external.cpython-35: module references __file__
python3.5.importlib.__pycache__._bootstrap_external.cpython-35: module references __path__
python3.5.importlib.__pycache__.util.cpython-35: module references __path__
python3.5.site-packages.pip.__pycache__.__init__.cpython-35: module references __file__
python3.5.site-packages.pip.__pycache__.locations.cpython-35: module references __file__
python3.5.site-packages.pip.__pycache__.__main__.cpython-35: module references __file__
python3.5.site-packages.pip._vendor.pkg_resources.__pycache__.__init__.cpython-35: module references __file__
python3.5.site-packages.pip._vendor.pkg_resources.__pycache__.__init__.cpython-35: module references __path__
python3.5.site-packages.pip._vendor.distlib.__pycache__.resources.cpython-35: module references __file__
python3.5.site-packages.pip._vendor.distlib.__pycache__.resources.cpython-35: module references __path__
python3.5.site-packages.pip._vendor.__pycache__.six.cpython-35: module references __path__
python3.5.site-packages.pip._vendor.__pycache__.__init__.cpython-35: module references __file__
python3.5.site-packages.pip._vendor.__pycache__.re-vendor.cpython-35: module references __file__
python3.5.site-packages.pip._vendor.requests.packages.urllib3.packages.__pycache__.six.cpython-35: module references __path__
python3.5.site-packages.pip._vendor.requests.__pycache__.certs.cpython-35: module references __file__
Traceback (most recent call last):
  File "setup.py", line 39, in <module>
    "mkmscimg=misoc.tools.mkmscimg:main",
  File "/usr/lib/python3.5/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.5/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.5/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 67, in run
    self.do_egg_install()
  File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "/usr/lib/python3.5/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.5/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/lib/python3/dist-packages/setuptools/command/bdist_egg.py", line 209, in run
    os.path.join(archive_root, 'EGG-INFO'), self.zip_safe()
  File "/usr/lib/python3/dist-packages/setuptools/command/bdist_egg.py", line 245, in zip_safe
    return analyze_egg(self.bdist_dir, self.stubs)
  File "/usr/lib/python3/dist-packages/setuptools/command/bdist_egg.py", line 355, in analyze_egg
    safe = scan_module(egg_dir, base, name, stubs) and safe
  File "/usr/lib/python3/dist-packages/setuptools/command/bdist_egg.py", line 392, in scan_module
    code = marshal.load(f)
ValueError: bad marshal data (unknown type code)

sbourdeauducq, 2017-03-21:
Those files should not be processed by setuptools. There are python2 sources that are part of compiler_rt and are not used by MiSoC; you can delete them to work around the problem, or figure out how to get setuptools to ignore them.

sbourdeauducq, 2017-07-13:
Actually the SyntaxError's on the python2 files are harmless and can be ignored. The real problem is this marshal data loading failure. Did you figure it out?