m-labs/misoc #52: error: Could not find a version that satisfies the requirement asyncserial (from miso c) (from versions: )

erickson-nist, 2017-07-13:
Copied from the pdq directory as requested. I have since also ran it in a python 3.5 environment and got the same error. Could you also please clarify which readme you updated? Looking at the history for the misoc and pdq readme fiels I don't see any recent commits.

As I mentioned in pdq issue 10, I am running into issues trying to compile the gateware. I was instructed to install ISE Webpack, but I do not believe that to be the problem. I have ISE Design Suite 14.7 with the free webpack license installed. The error comes in when attempting to install misoc (migen installs just fine).

I ran the following commands:
conda update conda
pip install -e git://github.com/m-labs/misoc.git#egg=misoc

That gives the following command line output:

(C:\ProgramData\Anaconda3) C:\Users\rabi\Documents\pdq-master>conda update conda

Fetching package metadata .................
Solving package specifications: .

*#* All requested packages already installed.
*#* packages in environment at C:\ProgramData\Anaconda3:
*#*
conda 4.3.22 py36_0 conda-forge/label/m
ain

(C:\ProgramData\Anaconda3) C:\Users\rabi\Documents\pdq-master>pip install -e git
://github.com/m-labs/misoc.git#egg=misoc
Obtaining misoc from git+git://github.com/m-labs/misoc.git#egg=misoc
Updating c:\users\rabi\documents\pdq-master\src\misoc clone
Requirement already satisfied: pyserial in c:\programdata\anaconda3\lib\site-pac
kages (from misoc)
Collecting asyncserial (from misoc)
Could not find a version that satisfies the requirement asyncserial (from miso
c) (from versions: )
No matching distribution found for asyncserial (from misoc)

sbourdeauducq, 2017-07-14:
Yes, asyncserial is not in pip. Install it via another means, or just install everything with conda (conda install misoc).

closed 2017-07-14 by sbourdeauducq

jordens, 2017-07-14:
Install asyncserial using pip: `pip install -e git://github.com/m-labs/asyncserial#egg=asyncserial`

mithro, 2017-07-15:
What is the reason you're not providing asyncserial via pip?

sbourdeauducq, 2017-07-15:
Conda is the package manager we support, and maintaining that one is enough hassle already.

mithro, 2017-07-15:
I really meant pypi rather than pip.

I was thinking about the possibility of using asyncserial outside the misoc project. It seems like the package would be generally useful to lots of people dealing with serial stuff.

sbourdeauducq, 2017-07-15:
Sure. ``conda install asyncserial`` works outside of misoc, or git clone + run ``setup.py`` manually. You may also setup and maintain another package installation system if you wish.