m-labs/misoc #55: RGMII on Sayma broken

jordens, 2017-09-01:
Should work but doesn't.

enjoy-digital, 2017-09-01:
@jordens: can you do a wireshark capture with the SoC directly connected to the computer? 

Just to understand what is going on: TX not working, TX working, RX not working?

If it's RX that is not working, can you desactivate hardware preamble and crc and add some trace in microudp to see what we receive? as its done here: 
https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/libnet/microudp.c#L372

If TX is not working, you can also desactivate hardware preamble and crc and add some traces in microudp to compare with what wireshark receive:
https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/libnet/microudp.c#L139

jordens, 2017-09-01:
When connected to a switch, that switch did no see any packets (ARP req, TX) coming in. RX was not tested IIRC. If we get Sayma hooked up (ping @whitequark ), we can try again. @sbourdeauducq was mostly working on this. I'll defer to him.

sbourdeauducq, 2017-09-01:
RX: definitely broken - no packets are received in the runtime at all.

TX: tested with the BIOS netboot and no packets were transmitted. But that's not a good test - since no one uses it regularly netboot could be bitrotten. Could not test in the runtime since it won't transmit anything before receiving a packet.

enjoy-digital, 2017-09-01:
@sbourdeauducq: avoid using a switch for debugging TX. Even if your packet is corrupted you'll be able to see it with Wireshark when connected directly to the computer.

jordens, 2017-09-01:
That depends on the ethernet adapter. And the switch did not show any corrupted packets either.
But yes, with some adapters you also get the currupted ones.

sbourdeauducq, 2017-09-01:
> RX was not tested IIRC

RX was tested, I modified the runtime to add a print whenever a packet was received, and nothing came (even though the LED on the switch for the SFP port was blinking).

mithro, 2017-09-06:
We had similar problem with the HDMI2USB firmware. Our current workaround is to use https://github.com/timvideos/HDMI2USB-litex-firmware/blob/master/gateware/rgmii_if.vhd and https://github.com/timvideos/HDMI2USB-litex-firmware/blob/master/gateware/s6rgmii.py

This is on a Spartan 6 rather than a Series 7 though.

sbourdeauducq, 2017-09-06:
The only significant difference I see is the delay elements. Maybe the data is simply corrupted and packets are dropped. Note that I rewrote significant parts of the PHY to make it portable.

closed 2018-10-10 by sbourdeauducq