m-labs/misoc #60: Papilio pro default rom size is too small

FelixVi, 2017-11-17:
In misoc/targets/papilio_pro.py:

```
self.register_rom(self.spiflash.bus)
```
 defaults to a rom size of 0xa000. This is too small for the current bios image and results in an overflow when software is compiled:

```
lm32-elf-ld: bios.elf section `.rodata' will not fit in region `rom'
lm32-elf-ld: region `rom' overflowed by 1156 bytes
```

The size can be adjusted by replacing above line with the following:
```
self.register_rom(self.spiflash.bus,rom_size=0xb000)
```

Are there any other steps needed after rom size has been adjusted?
How does the rom size interact with cpu_reset_address?
What does the variable "dummy" do in the SpiFlash class? Default is 15, papilio pro sets dummy to 4.

FelixVi, 2017-11-25:
Booting bios from BRAM works.
But even with larger ROM size, I had no luck getting flash boot to work.

Here's what I am using:
```
class BaseSoC(SoCSDRAM):
    def __init__(self, **kwargs):
        platform = papilio_pro.Platform()
        clk_freq = 80*1000000
        SoCSDRAM.__init__(self, platform, clk_freq,
                          cpu_reset_address=0x170000,
                          **kwargs)

        self.submodules.crg = _CRG(platform, clk_freq)

        self.submodules.sdrphy = GENSDRPHY(platform.request("sdram"))
        sdram_module = MT48LC4M16(clk_freq, "1:1")
        self.register_sdram(self.sdrphy, "minicon",
                            sdram_module.geom_settings, sdram_module.timing_settings)

        if not self.integrated_rom_size:
            self.submodules.spiflash = spi_flash.SpiFlash(platform.request("spiflash2x"),
                                                          dummy=4, div=6)
            self.flash_boot_address = 0x70000
            self.register_rom(self.spiflash.bus, 0x1000000)
            self.csr_devices.append("spiflash")

def main():
    parser = argparse.ArgumentParser(description="MiSoC port to the Papilio Pro")
    builder_args(parser)
    soc_sdram_args(parser)
    args = parser.parse_args()

    soc = BaseSoC(**soc_sdram_argdict(args))
    builder = Builder(soc, **builder_argdict(args))
    builder.build()
```

I am then uploading the top.bit and bios.bin files to the Papilio Pro using xc3sprog:

```
xc3sprog -c ftdi -R -I<bscan_file> <path>/top.bit:w:0x0:BIT <path>\bios.bin:w:0x70000:BIN
```

Monitoring the serial port, there is no bios prompt.

whitequark, 2017-11-25:
Does it work in simulation?

FelixVi, 2017-11-25:
Is there some notes on how simulation should be setup for misoc platforms?

I'm happy to run a couple tests, but wasn't able to find anything that'll help me get started.
Is there an easy way you guys use to turn builder output into testbenches?

sbourdeauducq, 2017-12-15:
No, we don't have a good simulator. You can try remapping the spiflash core to another address, boot the system with the integrated ROM, and then read from the flash using the new spiflash core address to check the contents as seen from the SoC.

FelixVi, 2018-11-14:
There seems to be a fix in above reference.

closed 2018-11-14 by FelixVi

keesj, 2019-01-30:
I can confirm that that booting from BRAM works.

python3 -m misoc.targets.papilio_pro --integrated-rom-size $((0x4500))

results in a bit file that I can flashed using

./papilio-prog -b bscan_spi_lx9.bit  -f misoc_basesoc_papilio_pro/gateware/top.bit
When changing those addresses on needs to do a clean build because the bios does not get recompiled when the start address changes.

Alternatively I made this small patch to default to using bram. 

```
diff --git a/misoc/targets/papilio_pro.py b/misoc/targets/papilio_pro.py
index 6c7a1a35..c32750b7 100755
--- a/misoc/targets/papilio_pro.py
+++ b/misoc/targets/papilio_pro.py
@@ -69,8 +69,8 @@ class BaseSoC(SoCSDRAM):
     def __init__(self, **kwargs):
         platform = papilio_pro.Platform()
         clk_freq = 80*1000000
-        SoCSDRAM.__init__(self, platform, clk_freq,
-                          cpu_reset_address=0x60000,
+        SoCSDRAM.__init__(self, platform, clk_freq,integrated_rom_size=0x4500,
+                          cpu_reset_address=0x0,
                           **kwargs)

         self.submodules.crg = _CRG(platform, clk_freq)
```

keesj, 2019-01-31:
> I am then uploading the top.bit and bios.bin files to the Papilio Pro using xc3sprog:
> 
> ```
> xc3sprog -c ftdi -R -I<bscan_file> <path>/top.bit:w:0x0:BIT <path>\bios.bin:w:0x70000:BIN
> ```
> 
> Monitoring the serial port, there is no bios prompt.
Right. This is something different. bios.bin should be loaded at the load address of the cpu_reset , not the flash boot address e.g. 0x170000 in your case. the flash_boot_address is the address that gets hardcoded into the bios to know where the load the next stage and the format of the image contains a header and crc.

I got the system working (I think) but this code still needs cleanup.

keesj, 2019-01-31:
Compared to $CURRENT master I have only the following changes.
```
index 146ae988..f115245d 100644
--- a/misoc/integration/soc_core.py
+++ b/misoc/integration/soc_core.py
@@ -12,7 +12,7 @@ __all__ = ["SoCCore", "soc_core_args", "soc_core_argdict"]

 class SoCCore(Module):
     mem_map = {
-        "rom":      0x00000000,
+        "rom":      0x00060000,
         "sram":     0x10000000,
         "main_ram": 0x40000000,
         "csr":      0x60000000,
@@ -135,9 +135,9 @@ class SoCCore(Module):

     def register_rom(self, interface, rom_size=0xa000):
         self.add_wb_slave(self.mem_map["rom"], rom_size, interface)
-        assert self.cpu_reset_address < rom_size
-        self.add_memory_region("rom", self.cpu_reset_address,
-                               rom_size-self.cpu_reset_address)
+        if not self.mem_map["rom"] <= self.cpu_reset_address < self.mem_map["rom"] + rom_size:
+            raise ValueError("CPU reset address 0x{:x} should be within ROM address range 0x{:x}-0x{:x}".format(self.cpu_reset_address,self.mem_map["rom"],self.mem_map["rom"] + rom_size ))
+        self.add_memory_region("rom", self.mem_map["rom"],rom_size)

     def get_memory_regions(self):
         return self._memory_regions

```
and I am able to flash the content using the following command

```
papilio-prog -r -b bscan_spi_lx9.bit  -f ./misoc_basesoc_papilio_pro/gateware/top.bit -a 60000:./misoc_basesoc_papilio_pro/software/bios/bios.bin
```

To test the code pressing reset is not enough I also need to unplug the board.