m-labs/misoc #61: Compiling gcc-7.2.0

FelixVi, 2017-11-17:
When compiling gcc-7.2.0, libquadmath was causing an error.

Not sure if there's anything more behind this, but I got past the problem by changing step 4 in the readme from

```
4. Compile and install GCC. Take gcc-core and gcc-g++ from GNU
   (version 4.5 or >=4.9).
   rm -rf libstdc++-v3
   mkdir build && cd build
   ../configure --target=lm32-elf --enable-languages="c,c++" --disable-libgcc \
     --disable-libssp
   make
   make install
```

to

```
4. Compile and install GCC. Take gcc-core and gcc-g++ from GNU
   (version 4.5 or >=4.9).
   rm -rf libstdc++-v3
   mkdir build && cd build
   ../configure --target=lm32-elf --enable-languages="c,c++" --disable-libgcc \
     --disable-libssp --disable-libquadmath
   make
   make install
```

If this is a good fix, an update in the readme might be a good thing.

sbourdeauducq, 2017-12-07:
Should be fine. Though I'd be surprised if new gcc still works for lm32 without other bugs, so make sure working versions can be compiled.

sbourdeauducq, 2017-12-07:
Or check 7.2 thoroughly...

FelixVi, 2017-12-07:
So far, I have not noticed any issues with 7.2, but to be safe I'll downgrade to version 4.9. Thanks!

closed 2017-12-07 by FelixVi

mithro, 2017-12-08:
We have been actively using GCC 5.4 on the HDMI2USB project with the lm32 and it has been working fine - our conda packages are here -> https://github.com/timvideos/conda-hdmi2usb-packages/blob/master/lm32/gcc-nostdc/meta.yaml
```
  fn: gcc-5.4.0.tar.bz2
  url: ftp://ftp.gnu.org/gnu/gcc/gcc-5.4.0/gcc-5.4.0.tar.bz2
  sha256: 608df76dec2d34de6558249d8af4cbee21eceddbcb580d666f7a5a583ca3303a
```

We don't however do anything advanced in our firmware code, so might not be hitting broken code paths. Do you know if there is a regression suite we can use to test if the lm32 is working correctly?

sbourdeauducq, 2017-12-09:
Doesn't GCC come with one?