m-labs/misoc #63: DDR3 broken on some Sayma boards

sbourdeauducq, 2017-12-06:
https://github.com/m-labs/sinara/issues/382#issuecomment-349025235

gkasprow, 2017-12-08:
I have Vivado test designs that I used to verify DDR3 operation.
Both memory banks tests were successful.
Is there something wrong in HW?
On some boards there was not mounted R105resistor that may cause memory controller failure during initialisation. It was reported [here](https://github.com/m-labs/sinara/issues/209#issuecomment-332620991)
This may cause an issue only with DDR2 controller. But Xilinx core worked fine since this sets impedance reference only on SDRAM control lines which do not need to calibrate its impedance.

enjoy-digital, 2017-12-08:
@gkasprow: this is not related to HW but to our PHY. I think this is now fixed and we will close this when validated on all sayma amc.

jbqubit, 2017-12-11:
I am compiling .bit from master now and will report if it works.

jbqubit, 2017-12-11:
Confirmed that Memtest passes on my Sayma_AMC using .bit built from master.

sbourdeauducq, 2017-12-13:
Also still works on sayma1 and sayma2 at m-labs. @enjoy-digital thanks for the rapid resolution.

closed 2017-12-13 by sbourdeauducq