m-labs/misoc #66: flterm: memory leak

jbqubit, 2017-12-14:
After running flterm on several ports to look for MiSoC BIOS messages my machine slows down. 

[image]

May be related to unplugging USB cable while continuing to run flterm.

whitequark, 2017-12-15:
Yes, I've also seen flterm to hang in a busy loop. It's something related to asyncio, and the easiest way to reproduce is to run an flterm instance in the background and then another on the same port.

jonaskeller, 2018-02-17:
Not sure if this is related, but running just a single instance on windows consumes about 8GB/h before eventually crashing:

	>python -m misoc.tools.flterm com4  1>>uart_corelog
	Traceback (most recent call last):
	  File "C:\Anaconda3\envs\artiq3.4\lib\runpy.py", line 193, in _run_module_as_main
		"__main__", mod_spec)
	  File "C:\Anaconda3\envs\artiq3.4\lib\runpy.py", line 85, in _run_code
		exec(code, run_globals)
	  File "C:\Anaconda3\envs\artiq3.4\lib\site-packages\misoc\tools\flterm.py", line 294, in <module>
		main()
	  File "C:\Anaconda3\envs\artiq3.4\lib\site-packages\misoc\tools\flterm.py", line 286, in main
		loop.run_until_complete(flterm.main_task)
	  File "C:\Anaconda3\envs\artiq3.4\lib\asyncio\base_events.py", line 467, in run_until_complete
		return future.result()
	  File "C:\Anaconda3\envs\artiq3.4\lib\asyncio\futures.py", line 294, in result
		raise self._exception
	  File "C:\Anaconda3\envs\artiq3.4\lib\asyncio\tasks.py", line 240, in _step
		result = coro.send(None)
	  File "C:\Anaconda3\envs\artiq3.4\lib\site-packages\misoc\tools\flterm.py", line 238, in main_coro
		await self.port.write(fs[1].result())
	  File "C:\Anaconda3\envs\artiq3.4\lib\asyncio\futures.py", line 294, in result
		raise self._exception
	  File "C:\Anaconda3\envs\artiq3.4\lib\asyncio\tasks.py", line 240, in _step
		result = coro.send(None)
	  File "C:\Anaconda3\envs\artiq3.4\lib\asyncio\queues.py", line 164, in get
		getter = self._loop.create_future()
	  File "C:\Anaconda3\envs\artiq3.4\lib\asyncio\base_events.py", line 275, in create_future
		return futures.Future(loop=self)
	  File "C:\Anaconda3\envs\artiq3.4\lib\asyncio\futures.py", line 171, in __init__
		self._loop = loop
	MemoryError

This is with misoc  0.9 py35_1+git2e2b7838

sbourdeauducq, 2018-02-27:
@jonaskeller You can use this as an alternative: http://store.chipkin.com/articles/using-putty-for-serial-com-connections-hyperterminal-replacement
Speed is 115200.

jonaskeller, 2018-02-27:
It still behaves the same on windows, so it looks like that's a separate bug.
It's not really an issue for me anymore though, thanks to the PuTTY alternative - thanks for pointing that out.

whitequark, 2018-02-27:
Hm, I might have misdiagnosed it.

sbourdeauducq, 2018-02-28:
I don't see how your commit could fix it.
1. it would be pretty hard to accumulate 8GB of serial data over a few hours, unless Python's ``bytes`` storage is extremely inefficient.
2. you are only resetting the buffer once when the magic string is found, then it accummulates again.

whitequark, 2018-02-28:
Yes, you're right.

commit 742386f5 (2018-02-27): flterm: don't leak memory.

Fixes #66.
closed 2018-03-04 by jordens