m-labs/misoc #80: Record.connect return value not used

tpwrules, 2018-06-10:
https://github.com/m-labs/misoc/blob/2436a68dcb24ef6624a6c9f86bdccb2f2632c158/misoc/interconnect/wishbone.py#L455-L456

I was browsing the source and it looks like the above line is a bug. The return value of `master.connect(slave)` is discarded so the connection doesn't happen. I imagine it was intended to be `self.comb += master.connect(slave)`, like in all the other places that function is used.

jordens, 2018-06-10:
Thanks. Looks correct. Wanna shoot a patch?

tpwrules, 2018-06-10:
Oh, I saw the contribution guidelines and it says you guys don't use pull requests. I didn't want to go through the hassle of submitting a patch on the mailing list. Feel free to make this change yourself.

whitequark, 2018-06-10:
We do accept pull request in misoc, the contribution guide is outdated.

closed 2018-06-11 by tpwrules