m-labs/misoc #92: fix or remove Ultrascale SDRAM init

sbourdeauducq, 2018-07-31:
Since the SDRAM initialization is no longer regularly tested (as the Rust bootloader is used instead), it has stopped working on Sayma.
Options are:
* remove the BIOS completely; this means removing LM32 since there is no usable Rust toolchain for it.
* remove Ultrascale SDRAM support from MiSoC
* fix the BIOS on Sayma

jbqubit, 2018-07-31:
I don't fully understand the implications of these different paths. Are you proposing switching LM32 to mor1kx? Where would Ultrascale SDRAM support be housed if not in MiSoC? What caused it to stop working?

sbourdeauducq, 2018-07-31:
ARTIQ 4 no longer uses the MiSoC bootloader, and Ultrascale development continued in the ARTIQ bootloader but the changes were not ported to the MiSoC bootloader.

jbqubit, 2018-07-31:
Since Sayma support is part of ARTIQ 4 it seems like Sayma ought to also use the ARTIQ 4 bootloader. What are the prospects for this port?

whitequark, 2018-07-31:
Sayma uses the ARTIQ 4 bootloader currently (and there's no particular reason I know for it to go back). The Ultrascale SDRAM code in MiSoC could be useful for other people who use MiSoC on Ultrascale, not that any necessarily exist.