m-labs/misoc #102: different clocking of differentiator and integrator

SingularitySurfer, 2020-09-18:
https://github.com/m-labs/misoc/blob/169dd32c42022140e541af0954be473e715785b9/misoc/cores/cic.py#L32-L35
Idea:
The cic is only stable when the integrators are exactly compensated by the differentiators. If there would be a stall of the input data the cic would go unstable here bc the integrators keep on going. But this might not be a problem as you "clock" all your filters from the output backwards so this situation can never occur right?

SingularitySurfer, 2020-09-18:
@jordens

jordens, 2020-09-18:
Yes. The idea is that this will only work if the rates are steady. Let's show that they are or aren't.

Proper simultaneous output and input handshaking seemed tricky. Certainly if you want to consider situations where you are actually draining and refilling the pipeline.

jordens, 2020-09-23:
Not an issue. It's stated in the docstring that the rates need to be maintained externally.

closed 2020-09-23 by jordens